SleipnirGroup/Choreo #56: Running in MacOS not working

CrispyBacon1999, 2023-07-14:
Whenever I try to run the program on MacOS (Intel Silicon), I'm getting issues with it finding the dylib for `/usr/local/lib/libTrajoptLib.dylib`. I get the error of
```
dyld[53276]: Library not loaded: @rpath/libTrajoptLib.dylib
  Referenced from: <A497B11A-EBBB-325A-9BD5-006564060FE6> /Users/bacon/dev/frc/UntitledWaypointEditor/src-tauri/target/debug/untitled-waypoint-editor
  Reason: tried: '/Users/bacon/dev/frc/UntitledWaypointEditor/src-tauri/target/debug/libTrajoptLib.dylib' (no such file), '/Users/bacon/dev/frc/UntitledWaypointEditor/src-tauri/target/debug/libTrajoptLib.dylib' (no such file), '/usr/local/lib/libTrajoptLib.dylib' (no such file), '/usr/lib/libTrajoptLib.dylib' (no such file, not in dyld cache)
```

Is there a step that's not included in the readme that I'm missing?

Note: I've tried setting `LD_LIBRARY_PATH` and it doesn't seem to affect anything.

jlbabilino, 2023-07-15:
You can copy all files matching the pattern `target/debug/build/trajoptlib-*/out/lib/*.dylib` to `target/debug/`. This is what you can do temporarily until we come up with a better solution of automatically finding the dependent dynamic libraries.

jlbabilino, 2023-07-15:
I haven’t put it in the README because ideally it shouldn’t be a required step, but maybe it should be there at least temporarily.

PR #57 (2023-07-18): Automatic TrajoptLib Resource Management
Features
- TrajoptLib dynamic library dependencies are copied into application bundle
- CI script to generate platform-specific installer bundles (`.exe`, `.dmg`, `.deb`) (closes #13)
- `npm install` is now the step that trajoptlib is built in and resources are collected
- `npm run tauri dev` automatically locates dynamic library dependencies (closes #56)
- `WebView2Loader.dll` is copied automatically with npm scripts (Related to the [GNU toolchain issue](https://github.com/tauri-apps/tauri/issues/875)

Notes
- `.msi` installers didn't copy resources fo [... cut]
closed 2023-07-18 by shueja-personal