SleipnirGroup/Choreo #68: Linux version installed with `.deb` fails to run

jlbabilino, 2023-07-20:
Opening Choreo:
```
./choreo: error loading shared libraries: libcasadi.so.3.7: cannot open shared object file: No such file or directory
```
So `choreo` is able to find `libTrajoptLib.so`, but it can't find `libcasadi.so.3.7`.

```
objdump -x libTrajoptLib.so |grep 'R.*PATH'
```
This yields the following:
```
RUNPATH              /home/runner/work/Choreo/Choreo/src-tauri/target/release/build/trajoptlib-c5c9b7a7279c9368/out/lib
```
There should be another rpath entry for the current directory, so `$ORIGIN/`.

This is likely an issue with TrajoptLib not adding the proper rpath entries to `libTrajoptLib.so`.

calcmogul, 2023-07-20:
Looks like it's already listed: https://github.com/SleipnirGroup/TrajoptLib/blob/main/cmake/modules/FetchCasADi.cmake#L68-L69
```cmake
      message(STATUS "Building for Linux x64")
      set(CMAKE_INSTALL_RPATH "${CMAKE_INSTALL_RPATH};$ORIGIN/../lib;$ORIGIN")
```

calcmogul, 2023-07-20:
I think it's from the RPATH settings in https://github.com/SleipnirGroup/TrajoptLib/blob/main/CMakeLists.txt not appending.

calcmogul, 2023-07-20:
Also, why is `CMAKE_INSTALL_RPATH` set twice with the same thing?

calcmogul, 2023-07-20:
See if this works?
```diff
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 31bd2ca..975df5f 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -18,17 +18,15 @@ set(CMAKE_SKIP_BUILD_RPATH FALSE)
 # installing)
 set(CMAKE_BUILD_WITH_INSTALL_RPATH FALSE)
 
-set(CMAKE_INSTALL_RPATH ${CMAKE_INSTALL_PREFIX}/lib)
-
-# add the automatically determined parts of the RPATH
-# which point to directories outside the build tree to the install RPATH
+# Add the automatically determined parts of the RPATH which point to directories
+# outside the build tree to the install RPATH
 set(CMAKE_INSTALL_RPATH_USE_LINK_PATH TRUE)
 
-# the RPATH to be used when installing, but only if it's not a system directory
+# The RPATH to be used when installing, but only if it's not a system directory
 list(FIND CMAKE_PLATFORM_IMPLICIT_LINK_DIRECTORIES "${CMAKE_INSTALL_PREFIX}/lib" isSystemDir)
 if ("${isSystemDir}" STREQUAL "-1")
-  set(CMAKE_INSTALL_RPATH "${CMAKE_INSTALL_PREFIX}/lib")
-endif("${isSystemDir}" STREQUAL "-1")
+  set(CMAKE_INSTALL_RPATH "${CMAKE_INSTALL_RPATH};${CMAKE_INSTALL_PREFIX}/lib")
+endif()
 
 # Set default build type to release with debug info (i.e. release mode
 # optimizations are performed, but debug info still exists).
```

calcmogul, 2023-07-20:
https://github.com/SleipnirGroup/TrajoptLib/pull/44

closed 2023-07-20 by jlbabilino