SleipnirGroup/Choreo #89: `npm run tauri dev` can't find libTrajoptLib.so

calcmogul, 2023-07-26:
I followed these steps:

1. Delete the `target` folder and any left over .so files
2. Run `npm install` from the repo root
3. Run `npm run tauri dev` from the repo root

```bash
    Finished dev [unoptimized + debuginfo] target(s) in 2m 30s
/home/tav/git/Choreo/src-tauri/target/debug/choreo: error while loading shared libraries: libTrajoptLib.so: cannot open shared object file: No such file or directory
```
Here's what's in my folders:
```bash
[tav@myriad Choreo]$ ls
icon        node_modules       public     spec       tsconfig.json
index.html  package-lock.json  README.md  src        tsconfig.node.json
LICENSE     package.json       scripts    src-tauri  vite.config.ts
```
```bash
[tav@myriad src-tauri]$ ls
build.rs                    libcasadi_nlpsol_ipopt.so      libTrajoptLib.so
Cargo.lock                  libcoinmetis.so.2              src
Cargo.toml                  libcoinmumps.so.3              target
icons                       libgfortran-040039e1.so.5.0.0  tauri.conf.in.json
libcasadi-tp-openblas.so.0  libipopt.so.3                  tauri.conf.json
libcasadi.so.3.7            libquadmath-96973f99.so.0.0.0
```

shueja-personal, 2023-07-29:
You might try `npm run tauri dev -- --release`. This fixes a similar error on Windows.

calcmogul, 2023-07-29:
The steps in the original post work for me now. :shrug:

calcmogul, 2024-02-21:
OBE by #326.

closed 2024-02-21 by calcmogul