SleipnirGroup/Choreo #92: Can't Generate Path

CrispyBacon1999, 2023-07-28:
After making a fairly long path, the generate button becomes disabled with the text "Generate path (needs 2 waypoints)". This is on the current alpha release version.

<details>
<summary>
<h3>Path:</h3>
</summary>

```json
{
    "version": "v0.0.0",
    "robotConfiguration": {
        "mass": 63.5029,
        "rotationalInertia": 5.6,
        "wheelMaxTorque": 1.9,
        "wheelMaxVelocity": 16,
        "wheelbase": 0.57785,
        "trackWidth": 0.57785,
        "bumperLength": 0.9,
        "bumperWidth": 0.9,
        "wheelRadius": 0.0508
    },
    "paths": {
        "Broken Path": {
            "waypoints": [
                {
                    "x": 1.847985029220581,
                    "y": 2.9409890174865723,
                    "heading": 3.1286062808745037,
                    "velocityMagnitude": 0,
                    "velocityAngle": 0,
                    "xConstrained": true,
                    "yConstrained": true,
                    "headingConstrained": true,
                    "velocityMagnitudeConstrained": false,
                    "velocityAngleConstrained": false,
                    "angularVelocity": 0,
                    "angularVelocityConstrained": false,
                    "controlIntervalCount": 0
                },
                {
                    "x": 2.8469085693359375,
                    "y": 3.332319974899292,
                    "heading": 3.141592653589793,
                    "velocityMagnitude": 0,
                    "velocityAngle": 0,
                    "xConstrained": true,
                    "yConstrained": true,
                    "headingConstrained": true,
                    "velocityMagnitudeConstrained": false,
                    "velocityAngleConstrained": false,
                    "angularVelocity": 0,
                    "angularVelocityConstrained": false,
                    "controlIntervalCount": 0
                },
                {
                    "x": 5.946660995483398,
                    "y": 3.414705514907837,
                    "heading": 0,
                    "velocityMagnitude": 0,
                    "velocityAngle": 0,
                    "xConstrained": true,
                    "yConstrained": true,
                    "headingConstrained": true,
                    "velocityMagnitudeConstrained": false,
                    "velocityAngleConstrained": false,
                    "angularVelocity": 0,
                    "angularVelocityConstrained": false,
                    "controlIntervalCount": 0
                },
                {
[... 176 lines cut ...]
                    "angularVelocityConstrained": false,
                    "controlIntervalCount": 0
                },
                {
                    "x": 2.0333521366119385,
                    "y": 3.476494550704956,
                    "heading": 3.0118902024990373,
                    "velocityMagnitude": 0,
                    "velocityAngle": 0,
                    "xConstrained": true,
                    "yConstrained": true,
                    "headingConstrained": true,
                    "velocityMagnitudeConstrained": false,
                    "velocityAngleConstrained": false,
                    "angularVelocity": 0,
                    "angularVelocityConstrained": false,
                    "controlIntervalCount": 0
                }
            ],
            "trajectory": null
        }
    }
}
```
</details>

CrispyBacon1999, 2023-07-28:
Did some digging into the code and it looks like it just renders that way, even while it's generating. Eventually, the button did go back to the enabled state, but it still doesn't generate a path for it.

jlbabilino, 2023-07-28:
Correct. The new release has a hamster wheel progress indicator to show this.

shueja-personal, 2023-07-29:
As far as the generation failing, we know the solver struggles with longer paths. We have some room for improvement with it.

closed 2023-08-03 by shueja-personal