SleipnirGroup/Choreo #96: Math expression parsing for numerical entry text boxes

jlbabilino, 2023-07-31:
Example: typing `2 + 3` into a number text box should enter the value of `5`.

Basic mathematical syntax, functions, and constants should be supported. `(sin(pi/2)+1*2)^2` will result in `9`.

An example implementation of this is in [HelixNavigator](https://github.com/TripleHelixProgramming/HelixNavigator/blob/master/app/src/main/java/org/team2363/lib/math/ExpressionParser.java).

jlmcmchl, 2023-07-31:
Is this intended to support limits? ie sin(0)/0 is undefined on its own, but lim x->0 sin(x)/x is 1.

jlbabilino, 2023-08-02:
Is there a specific situation you need that kind of functionality?

jlbabilino, 2023-08-06:
I think [mathjs](https://mathjs.org/index.html) is the perfect candidate for this.

PR #562 (2024-08-27): Rearchitect for 2025
* expression parsing instead of number inputs
* new file schema
* redesigned constraints system

Closes #565 
- [X] ui
- [x] trajoptlib

Closes #96 

Closes #566
- [X] file schema
- [X] backend storage of variables

(UI panel to add/edit variables is separate issue)

Closes #568

Closes #569 

Closes #565
closed 2024-08-27 by shueja