SleipnirGroup/Choreo #110: App crash with long path

Ashray-g, 2023-08-08:
[image]
[image]

Path JSON:
https://pastebin.com/4UDPUzn3

[image]

Using arm release 0.2.0

Awesomeplayer165, 2023-09-15:
from discord,

[image]

Awesomeplayer165, 2023-12-04:
for now, I decided its best to reduce the number of control intervals to around 20 or so. Click gear in your path (sidebar), uncheck guess path detail, type replace 40 with 20.

calcmogul, 2024-03-03:
OBE by #326.

closed 2024-03-03 by calcmogul