SleipnirGroup/Choreo #115: Dalton's Feedback

Daltz333, 2023-09-19:
I didn't actually check if any issues are already open, but this is a collection of my feedback.

# Feature Requests

- Custom field image

# Bugs
- No way to close "Robot Config" after clicking on it.
- Clicking off of a path name should save the name (I didn't realize I needed to press "enter" to save the path)
- After deleting all waypoints, I was unable to add any?

[image]

- Memory leak when adding paths. Added 100 paths, initially spiked to 200MB then settled down to 100. Was initially 80. I can reliably reproduce this and have gotten it to about 300MB settled after ~300 paths. 

# General Things

- Application started very tiny (maybe like 600x600px)
- Click area for an existing waypoint is too tiny
- Opening an invalid JSON doesn't do anything (my shrek movie as JSON passed), but it should at least tell the user it's bad.
- Creating a new file and attempting to export a trajectory (no points created), does nothing.
  - In general, a couple more things need user feedback (notifying the user something is happening)
  - e.g. generating trajectory, 
- Tooltips for Mass/Mol/Robot constants should be added
- Choreo version should be added somewhere, maybe titlebar of the application (for user debugging, I just pulled the artifact from 
53be0fa.
- Launches with a cmdline, that should probably be hidden
- Name of the app in task manager seems incorrect 
[image]

# Overall thoughts

Application is quite stable for me. Wasn't able to force a crash without doing malicious things (deleting app/system files). My complains are honestly pretty minor.

jlbabilino, 2023-09-19:
Thanks so much for taking the time to collect all this information!

calcmogul, 2024-03-03:
The issues mentioned here are now tracked by subissues.

closed 2024-03-03 by calcmogul