SleipnirGroup/Choreo #137: Unable to build on Windows due to call to non-existent 'rm'

willtoth, 2023-10-27:
When trying to build following the instructions on the wiki, I get the following error

```
Building trajoptlib dylibs
'rm' is not recognized as an internal or external command,
operable program or batch file.
node:child_process:965
    throw err;
    ^

Error: Command failed: cd C:\Users\wtoth\Src\Choreo\scripts/../src-tauri && rm -rf target && cargo build --release
'rm' is not recognized as an internal or external command,
operable program or batch file.

    at checkExecSyncError (node:child_process:890:11)
    at execSync (node:child_process:962:15)
    at Object.<anonymous> (C:\Users\wtoth\Src\Choreo\scripts\copy_trajoptlib_dylibs.cjs:29:1)
    at Module._compile (node:internal/modules/cjs/loader:1376:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1435:10)
    at Module.load (node:internal/modules/cjs/loader:1207:32)
    at Module._load (node:internal/modules/cjs/loader:1023:12)
    at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:135:12)
    at node:internal/main/run_main_module:28:49 {
```

It looks like the copy script is making calls to Windows cmd that are not cross platform.

agostinopique, 2023-10-27:
Could you link the wiki you're following?

willtoth, 2023-10-27:
> Could you link the wiki you're following?

https://github.com/SleipnirGroup/Choreo/wiki/Building-Choreo

calcmogul, 2024-02-20:
OBE by #326.

closed 2024-02-20 by calcmogul