SleipnirGroup/Choreo #139: Progress bar instead of spinning wheel

willtoth, 2023-10-27:
This is a feature request. When generating a path, sometimes it takes quite some time. Is there a way to show a progress bar, or at least some kind of progress indicator beyond just the spinning wheel of death. The spinning wheel is certainly an improvement over the previous, but it still is hard to wait and know if something is happening.

jlbabilino, 2023-10-27:
As far as we can tell, the only feedback the optimizer (CasADi + IPOPT) gives us is the current iteration number. Unfortunately, we can’t know for sure how many iterations it will take to reach the solution, but in my opinion the best way a progress bar could be created would be to represent progress with `current_iter / max_iter`. Then it may or may not reach the end, but it gives you an idea that it’s making progress.

calcmogul, 2024-02-24:
We can get a better estimate of the total number of iterations to reach the 1e-8 error tolerance with this: https://en.wikipedia.org/wiki/Interior-point_method#Convergence_and_complexity

shueja, 2024-04-10:
Superseded by solver progress feedback #439

closed 2024-04-10 by shueja