SleipnirGroup/Choreo #141: Crashes don't produce useful feedback

willtoth, 2023-11-12:
I build the app off of the latest main branch. When generating the path, the whole app crashes. However I don't have good data to provide for this bug ticket, as the crash produces no useful information.

Lewis-Seiden, 2023-11-12:
do you have a file with the path that crashed the app?

willtoth, 2023-11-12:
I don't. For whatever reason, _any_ three points followed by generating the path, would crash the app.

shueja, 2023-11-13:
* Would you provide your system information (os, etc)?
* How were you running the app?

We're investigating a potentially related crash bug that crashes even with two points. Do you know if it required at least 3 points to crash it?

willtoth, 2023-11-13:
- System Specs (let me know if I'm missing anything)
  - Windows 11
  - npm 10.2.0
  - node 21.1.0
  - cargo 1.73.0
  - rustup default `stable-x86_64-pc-windows-gnu`
- Built it from c32aa0a0720777626b70a208bf9803d09f25c7bc, installed using nsis installer _or_ run using `npm run tauri dev -- --release`
- Any number of points except 1 since you can't generate with 1

Note I did not see this when I built a few weeks ago.

I am trying to add some debugging to the application, but this requires a bit of learning on my part.

shueja, 2023-11-14:
This sounds like the issue that broke our Windows CI builds sometime between September 10 and October 16. From what we can tell, it's a build environment change (GH updated the Windows runner, and LLVM released major version 17, and we're not sure which change actually broke it yet). Would you try the Windows installer from the v0.3.0 release? That was built outside of the CI pipeline due to this issue, and has been tested to not crash.

Also, would you please paste the output of `clang --version`? We built v0.3.0 on my machine with LLVM/Clang 16.0.6, but CI uses the latest, 17.0.3.

willtoth, 2023-11-14:
```
clang version 17.0.1
Target: x86_64-pc-windows-msvc
Thread model: posix
InstalledDir: C:\Program Files\LLVM\bin
```

Note that I did not have this issue a week ago, and afaik my LLVM/Clang setup has not changed in that time. Hope its not some kind of flaky issue with the compiler.

calcmogul, 2024-03-03:
OBE by #326.

closed 2024-03-03 by calcmogul