SleipnirGroup/Choreo #184: Static obstacles support on GUI

lidor51, 2023-12-03:
Support generating a path, using a static obstacles constraints in the GUI.

Lewis-Seiden, 2023-12-03:
the problem with a fixed set of obstacles is that a full field of obstacles greatly slows down generation with the current formulation. Adding custom ones/toggling presets is very doable however

lidor51, 2023-12-03:
Obviously full fields obstacles + custom is the best solution, but if it's effecting generation time that much, I guess that cutsom ones are ok.

If the user can define them per field and toggle the necessary obstacles per path it would be nice. So you don't need to create them again per path.
(Additionaly if Choreo would have also a set of predefined separated obstacles per field it can be nice).