SleipnirGroup/Choreo #210: Initial Guesses on Translation Waypoints don't respect optimal wrapping.

jlmcmchl, 2023-12-30:
See title - in [this sample project](https://gist.github.com/jlmcmchl/cc164a7ad3ddb6e808d2f68d4a27622f), the generated trajectory does a pair of unnecessary rotations in order to reach an initial guess waypoint that instead should have been shorter rotations.

Expected result:

https://github.com/SleipnirGroup/Choreo/assets/1332080/6ed81545-3cc3-4998-909f-ce69531c114e

Actual: 

https://github.com/SleipnirGroup/Choreo/assets/1332080/b9c6213e-24f7-44cd-b1f5-20d59723b8f3

Awesomeplayer165, 2024-01-02:
can you send your choreo file for actual? my guess is that you are wrapping neg pi and pos pi.

Awesomeplayer165, 2024-01-02:
i answered you on discord. did this answer your question?

jlmcmchl, 2024-01-02:
I'd like to keep this issue open - it's still a problem that should be tracked. For most paths, the workaround is just choosing a heading on translation waypoints that's on the correct half of the +/- heading boundary.

shueja, 2024-01-09:
https://github.com/SleipnirGroup/Choreo/assets/32416547/69154ea0-127a-4d4d-929c-505720367b4d

I drew a similar path in the released 2024.1.0. Am I doing something incorrect in reproducing the problem, or does it seem solved on your end as well?

shueja, 2024-08-25:
Solved by using manifold logic in TrajoptLib

closed 2024-08-25 by shueja