SleipnirGroup/Choreo #243: bug: origin point on 2024 field image is off

rgodha24, 2024-01-10:
[image]

I'm on MacOS x86, both with the official release build and a local version. I think it's a Mac only (eg. Safari webkit) issue because the image in the CD post (duplicated below) is fine

[image]

Awesomeplayer165, 2024-01-10:
I can confirm it happens on ARM64 and Rosetta 2

rgodha24, 2024-01-10:
it seems to be the image that's off, because the lines are still at 0,0

[image]

rgodha24, 2024-01-10:
i found a fix, working on the PR rn

shueja, 2024-01-15:
Closed by #246

closed 2024-01-15 by shueja