SleipnirGroup/Choreo #270: [Feature Request] Duplicate/copy waypoints

GalexY727, 2024-01-17:
piggybacking off of #249, I think it would be very useful to be able to copy a waypoint and paste/duplicate it into another path in the case that there is a path that goes to pick up a note (ends there) and another one chains off of it to either come back and shoot in the speaker or place in the amp, for example. Both paths would want to start at the same location, and it would be easiest to do that through a copy/paste mechanism. I assume if #249 is completed this shouldn't be much different.

I would love to try to tackle this if I am given a place to start :)

GalexY727, 2024-01-17:
I would say using the grid but that isn't necessarily where you want to end your autos.. it's not modular enough

GalexY727, 2024-01-17:
The start/end points of an auto path could go into a list of "snappable" positions, and it loops through the list to see if the current suggestion is near that path....

GalexY727, 2024-01-17:
> one idea ive had that solves the same/similar problem is being able to snap ends of paths to each other somehow

oh! what if there was a "fork path start | end" button which made another path with a waypoint pre-generated at the start/end of the path

Lewis-Seiden, 2024-01-17:
> "fork path (start | end)" button

i like this as a simpler version of the feature, although prone to issues when paths are edited later and fall out of sync. a path sync system shouldnt be awful (?) to implement, and is better long term imo. if you want to take a pass at copying paths that shouldnt be too hard though

PatribotsProgramming, 2024-02-06:
Pathplanner has a "link waypoint" system where you can have a single name for a waypoint that links with the rest of the project. Rather than linking the whole path, if this feature was on a waypoint-to-waypoint scope then it might be more accomplishable. Something to do with a global hash map...

shueja, 2024-03-05:
Original request implemented for 2024.1.3