SleipnirGroup/Choreo #282: Windows Defender Triggered

otter502, 2024-01-22:
I installed the most recent version of Choreo from this github repo, installed it, and it failed to run with the error of "file contains virus or unwanted software". I went through my windows defender logs and it said it detected "Trojan:Script/Wacatac.B!ml"
I compared the md5 hash (copied below) to a friend who didn't have this problem and it was the same. I also sent the exe (not the installer) through virus total before windows defender quarantined and deleted it. (linked below as well, on the details page is a bunch of hashes)

md5 hash: `d39c69f4624c8998b5d540d4b18439b2`
VT page: https://www.virustotal.com/gui/file/b15188142a0645222be495d2c4dd5aae0a583f6d11cba24e2d01403fd3dc1ee2/detection

calcmogul, 2024-01-22:
Seems like a false positive. Signing the installer can apparently help avoid it.

gerth2, 2024-01-23:
For continuity, same symptom on my machines - desktop is fine, laptop flags it, same release and both with windows 11

calcmogul, 2024-03-03:
We can't realistically fix this on our end.

closed 2024-03-03 by calcmogul