SleipnirGroup/Choreo #283: [releases] add an appimage or a non-deb package

OnlyCS, 2024-01-22:
Hey,

I'm using an x64 based linux distro but the deb file in releases will not work with my Fedora system. Any change of switching to a unified packaging system, like a flatpak, AppImage, or supporting builds for these other distributions?

For now, I will build this project from source.

Thanks for your help

calcmogul, 2024-01-22:
Here's the docs for Tauri packaging, for reference: https://tauri.app/v1/guides/building/linux/.

Looks like the only other option is AppImage.

OnlyCS, 2024-01-22:
welp looks like i cant build on fedora due to lack of libsoup @ 2.4 (i have 2.7.4, no 2.4.x on fedora repos)

OnlyCS, 2024-01-23:
whoops found a fix. running an appimage here! ill make a pr
[image]

calcmogul, 2024-02-21:
Now that we link everything statically, there shouldn't be missing libraries anymore. It's worth trying again.

OnlyCS, 2024-02-29:
ill take a look at this in a bit

PR #370 (2024-03-04): Add AppImage build
Fixes #283.
commit 4eee3151 (2024-03-04): Add AppImage build (#370)

Fixes #283.
closed 2024-03-04 by calcmogul