SleipnirGroup/Choreo #290: [Feature Request] ChoreoTrajectory.getStopPointCount()

GalexY727, 2024-01-27:
With the added feature of splitting trajectories by their stop points on export, it might be useful (and easy to implement?) to have a method that returns the number of stop points for a given path to use in things like for-loops. 

I had originally thought it would just be checking how many states have an x and y velocity of zero, but then I realized that there could be a pseudo-stop point that has the robot flip direction without stopping which might result in a false positive.

I would be okay with tackling this if it's that simple 🙂.

GalexY727, 2024-01-27:
A more optimized approach would be finding a way to get the count [as it is loaded](https://github.com/SleipnirGroup/Choreo/blob/main/choreolib/src/main/java/com/choreo/lib/Choreo.java#L78) but that sounds like it might be a bit more complicated...

shueja, 2024-01-28:
The most surefire way you'd be able to do this with the current exports is to read the directory and look for the segment files while loading the full path. I'm not sure of a use-case where you'd need the number of stop points, and knowing the usecase will determine the best solution.

GalexY727, 2024-02-01:
The specific use case I had in mind was to be able to preload all paths `from pathplanner, while using the "split traj by stop points" option` when the robot started so that it is ready to run the moment teleop starts. If it takes no time at all to load a path once autonomous starts, then this issue would be pretty invalidated.

shueja, 2024-02-01:
So you're looking to get the names of the split parts for each segment in order to load them with PathPlannerLib?

GalexY727, 2024-02-02:
Not necessarily the names, although that would also be useful. Here's some psudo-code for this use case:
```java
String name = "Path";
PPPath[] list;
for (int i = 1; i < ChoreoTrajectory.getStopPointCount(name); i++) {
  list.add(PP.loadChoreoTraj(name+"."+i);
}
```

GalexY727, 2024-02-02:
Again, it's a very specific use case and it might not prompt the creation of an entire feature.

shueja, 2024-09-02:
This is obsolete for 2025.