SleipnirGroup/Choreo #308: Binary from Windows CI crashes with double-free

calcmogul, 2024-02-01:
There was a change in the runner image from https://github.com/actions/runner-images/blob/win22/20230910.1/images/win/Windows2022-Readme.md to https://github.com/actions/runner-images/blob/win22/20231002.1/images/win/Windows2022-Readme.md. With zero code changes (i.e. rerunning the exact same commit), the builds started crashing on Windows.

TrajoptLib uses CasADi 3.6.4 from November 2023.

shueja, 2024-02-01:
https://github.com/actions/runner-images/issues/8266 This seems to be a report of others having issues with the upgrade from gcc 11.2.0 (listed as `Mingw-w64 11.2.0` in the older changelog) to gcc 12.2.0.

calcmogul, 2024-02-01:
GDB backtrace from @shueja:
[choreo_gdb_dump.txt](https://github.com/SleipnirGroup/Choreo/files/14130969/choreo_gdb_dump.txt)

calcmogul, 2024-02-07:
Fixed by #313.

closed 2024-02-07 by calcmogul