SleipnirGroup/Choreo #343: Quit Unexpectedly When Generating Path With 13 Waypoints

gonzalezgonzalezl, 2024-02-24:
I was trying to create a path with 13 waypoints, and when I clicked on the generate button the application closed, and a message popped up saying that Choreo quit unexpectedly. I tried this multiple times and the same thing kept happening. I'm using an M1 chip Mac, version 14.2.1 and Choreo version 2024.1.3.

Here is (some of the) problem report from the "Choreo Quit Unexpectedly" message:
<details>
Process:               Choreo [74966]
Path:                  /Applications/Choreo.app/Contents/MacOS/Choreo
Identifier:            org.sleipnirgroup
Version:               2024.1.3 (20240206.065303)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-02-23 18:02:27.8910 -0800
OS Version:            macOS 14.2.1 (23C71)
Report Version:        12
Anonymous UUID:        2DDAE16B-5A7D-7F84-8684-2166EF7B6985

Sleep/Wake UUID:       D5118719-83AE-45C0-B76B-019468052584

Time Awake Since Boot: 1700000 seconds
Time Since Wake:       4984 seconds

System Integrity Protection: enabled

Crashed Thread:        4  tokio-runtime-worker

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 6 Abort trap: 6
Terminating Process:   Choreo [74966]

Application Specific Information:
abort() called

Thread 0:: main Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x1886e1874 mach_msg2_trap + 8
1   libsystem_kernel.dylib        	       0x1886f3cf0 mach_msg2_internal + 80
2   libsystem_kernel.dylib        	       0x1886ea4b0 mach_msg_overwrite + 476
3   libsystem_kernel.dylib        	       0x1886e1bf8 mach_msg + 24
4   CoreFoundation                	       0x1887ffbf4 __CFRunLoopServiceMachPort + 160
5   CoreFoundation                	       0x1887fe4bc __CFRunLoopRun + 1208
6   CoreFoundation                	       0x1887fd9ac CFRunLoopRunSpecific + 608
7   HIToolbox                     	       0x192dac448 RunCurrentEventLoopInMode + 292
8   HIToolbox                     	       0x192dac284 ReceiveNextEventCommon + 648
9   HIToolbox                     	       0x192dabfdc _BlockUntilNextEventMatchingListInModeWithFilter + 76
10  AppKit                        	       0x18bfda8a4 _DPSNextEvent + 660
11  AppKit                        	       0x18c7b4980 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 716
12  AppKit                        	       0x18bfcdd50 -[NSApplication run] + 476
13  Choreo                        	       0x100a2d098 objc_exception::try_no_ret::try_objc_execute_closure::hfcf61b62c462aa16 + 44
14  Choreo                        	       0x100b810e4 RustObjCExceptionTryCatch + 28
15  Choreo                        	       0x100a3d860 objc::message::platform::send_unverified::h84198cecbc799dbc + 92
16  Choreo                        	       0x100a9764c tao::platform_impl::platform::event_loop::EventLoop$LT$T$GT$::run::h4e64d21446f3a733 + 832
17  Choreo                        	       0x1009ed384 _$LT$tauri_runtime_wry..Wry$LT$T$GT$$u20$as$u20$tauri_runtime..Runtime$LT$T$GT$$GT$::run::h81c1b018f4b92361 + 240
18  Choreo                        	       0x100a42fcc tauri::app::Builder$LT$R$GT$::run::hd864c09ff4875f70 + 7600
19  Choreo                        	       0x100a18154 choreo::main::h1f478fad1a7394b2 + 4260
20  Choreo                        	       0x100a51c1c std::sys_common::backtrace::__rust_begin_short_backtrace::h4f67fdedfe156b9a + 12
21  Choreo                        	       0x100a8cb70 std::rt::lang_start::_$u7b$$u7b$closure$u7d$$u7d$::h6bf565bc2012a5bc + 16
22  Choreo                        	       0x100c91264 std::rt::lang_start_internal::h5f00c09289a72e67 + 648
23  Choreo                        	       0x100a194c4 main + 52
24  dyld                          	       0x1883a10e0 start + 2360

Thread 1:: com.apple.coreanimation.render-server
0   libsystem_kernel.dylib        	       0x1886e1874 mach_msg2_trap + 8
1   libsystem_kernel.dylib        	       0x1886f3cf0 mach_msg2_internal + 80
2   libsystem_kernel.dylib        	       0x1886ea4b0 mach_msg_overwrite + 476
3   libsystem_kernel.dylib        	       0x1886e1bf8 mach_msg + 24
4   QuartzCore                    	       0x190905fa4 CA::Render::Server::server_thread(void*) + 544
5   QuartzCore                    	       0x190905d74 thread_fun(void*) + 32
6   libsystem_pthread.dylib       	       0x188722034 _pthread_start + 136
[... 288 lines cut ...]
__OBJC_RO                         71.1M        1 
__OBJC_RW                         2168K        1 
__TEXT                           789.9M      686 
dyld private memory                272K        2 
mapped file                      252.2M       48 
shared memory                      864K       14 
===========                     =======  ======= 
TOTAL                              6.8G     3772 
TOTAL, minus reserved VM space     2.9G     3772 

...

Model: MacBookAir10,1, BootROM 10151.61.4, proc 8:4:4 processors, 8 GB, SMC 
Graphics: Apple M1, Apple M1, Built-In
Display: Color LCD, 2560 x 1600 Retina, Main, MirrorOff, Online
Memory Module: LPDDR4, Hynix
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4378), wl0: Aug 26 2023 17:55:53 version 18.20.439.0.7.8.163 FWID 01-f9b9247b
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB31Bus
Thunderbolt Bus: MacBook Air, Apple Inc.
Thunderbolt Bus: MacBook Air, Apple Inc.
</details>

calcmogul, 2024-02-24:
Could you upload the .chor file you used to produce the crash?

gonzalezgonzalezl, 2024-02-24:
[path.chor.json](https://github.com/SleipnirGroup/Choreo/files/14391501/path.chor.json)

calcmogul, 2024-03-03:
OBE by #326.

closed 2024-03-03 by calcmogul