SleipnirGroup/Choreo #354: Lack of feedback when receiving an infeasible error

Eternal2008, 2024-02-29:
When generating new path it returns an infeasible error. It has an unknown trigger and happens seemingly at random even on newly created paths with just 2 waypoints and nothing else.

calcmogul, 2024-02-29:
These spurious infeasible trajectory errors have been fixed on main by switching to a different optimizer backend. Please try one of the builds here: https://github.com/SleipnirGroup/Choreo/actions/runs/8084712732

Eternal2008, 2024-02-29:
Thanks for the suggestion now it says Bad file reference when attempting to generate the paths for any new paths I create, but not for any old ones I created. Those I can just regenerate without any issues.

calcmogul, 2024-02-29:
It may have been caused by a recent upgrade to the fs package. Try this build once it's finished: https://github.com/calcmogul/Choreo/actions/runs/8089723596

Eternal2008, 2024-02-29:
Ok now the first generation i run causes this 
[image]
and upon re attempting generation it goes back to saying their is a bad file descriptor
[image]

calcmogul, 2024-03-01:
This error message is from fmtlib during a write failure. I'll work on a patch for Sleipnir.

PR #361 (2024-03-01): Upgrade TrajoptLib to squelch write failure exceptions
Fixes #354.
closed 2024-03-01 by calcmogul