SleipnirGroup/Choreo #365: Enhance GUI Angle Setting with Support for Constants and Units

ori-coval, 2024-03-02:
## Description:
Currently, the GUI angle setting feature restricts users to input angles solely in Radians, lacking the flexibility to utilize mathematical constants such as π or employ alternative measurement units like degrees. 
This limitation can be cumbersome for users accustomed to degrees or those requiring precise mathematical constants in their work.

## Proposal:

Support for Mathematical Constants: Introduce support for common mathematical constants such as π. This would allow users to input angles using standard mathematical notation, enhancing usability and reducing cognitive load.

Flexibility in Measurement Units: Enable users to choose between different measurement units for angles, such as radians or degrees. This would accommodate diverse user preferences and streamline workflows for individuals accustomed to specific units.

jlbabilino, 2024-03-03:
Related to #96

shueja, 2024-08-25:
Duplicate of #96

closed 2024-08-25 by shueja