SleipnirGroup/Choreo #396: Choreo File Won't Open

AADad578, 2024-03-15:
We are having an issue where the attached choreo file will not open. It is giving an error seen below. This occurred after attempting to duplicate a path. It did not work, so we restarted it and it has been giving the error since. 

```
Opening file error: Internal parsing error: Error:[mbox-state-tree] No matching type for union ("first" | "last" | {uuid: string})
```

[choreo file](https://team2869.org/wp-content/uploads/2024/03/choero.txt)

calcmogul, 2024-03-15:
Same file hosted on GitHub:
[internal-parsing-error.chor.txt](https://github.com/SleipnirGroup/Choreo/files/14621570/internal-parsing-error.chor.txt)

Jared341, 2024-03-16:
The problem is that the duplicated choreo path contained a marker associated to a waypoint that was deleted in the duplicate path. Choreo chokes when a marker’s `target` field (i.e. waypoint index) is deserialized as null.

Manually removing the marker with `"target": null` from the .chor file fixed this for us.

closed 2024-03-19 by calcmogul