SleipnirGroup/Choreo #401: Opening Chor file causes all Traj Event Markers to be set to []

changsolomon8, 2024-03-19:
This might be a product of attempting to use Choreo in the context of a git repository with multiple individual computers changing paths (and then just checking in a **subset of Traj** files).

Currently, with our set up, if you open our shared .chor file, it immediately touches N different trajectory files and clears out the eventmakers to be [].  This causes headaches because we then have to manually stash the changes and only check in a subset of the Traj files that we want to change.  We intentionally do not want to regenerate some Trajectories.  

Is there a better way/recommended way to use .chor files in git repository? Or is the recommendation to just regenerate all the Trajectories on each commit type of workflow?

shueja, 2024-03-22:
Seems like there's two issues here:
1. Trajectories all export upon opening the file, when this is not desired
2. When exporting trajectories upon opening the file, the event marker array is empty when there are properly-generated event markers that should be there.
Is the second an accurate description of what you experience?

changsolomon8, 2024-03-22:
Yes, the second is an accurate description here. 

I also replied a little to Adam, but regarding automatically exporting upon opening file: 

> Yeah I’m not advocating for footgunning other teams. Just trying to understand if there something we can do here to make this easier to use.  I suspect that if we don't fix this we will end up shooting ourselves in the food because we accidently deploy eventmarkers as [].

shueja, 2024-03-26:
Fixed in v2024.2.2

closed 2024-03-26 by shueja