SleipnirGroup/Choreo #443: Show where generation error is

Tom-ne, 2024-04-08:
Currently, when there is an error generating the path, there is no way for the user to know where is the problem, I think it would be helpful the add to the error message shown to the user, where is the error in the path.

calcmogul, 2024-04-08:
Related to https://github.com/SleipnirGroup/TrajoptLib/issues/56.

This is deceptively difficult. For an infeasible trajectory, the best you can do is determine which constraints are violated, map the constraints to specific things like pose or dynamics, then report that to the user. Other error messages are solver internal errors we can't diagnose.

Tom-ne, 2024-04-08:
I think showing the user what constraints are violated will still speed up the learning process for new users and help find why is there an issue