SleipnirGroup/Choreo #458: When splitting trajectories at stop points, solve the pieces in parallel for faster solves

calcmogul, 2024-04-23:
Each piece would be solved on a separate thread, then merged once all are complete.

laviRZ, 2024-07-08:
Not only would this enable multithreading, but it would greatly simplify the problem given to the solver. Currently when playing with Choreo I find that as the path gets longer the solve time gets exponentially longer, making very long and complicated paths, especially when using obstacles, unsolvable most of the time. tl;dr would love to see this implemented.

calcmogul, 2024-07-08:
> it would greatly simplify the problem given to the solver. Currently when playing with Choreo I find that as the path gets longer the solve time gets exponentially longer, making very long and complicated paths

This is false. The problem's dimensionality alone has no effect on the problem's difficulty. Furthermore, the time per iteration only increases linearly for direct transcription problems, and the number of iterations to solve is the same if you don't invoke feasibility restoration (which should be the case for well-formulated problems).

> especially when using obstacles

This is what causes the solve time to drastically increase. Obstacles are a beta feature that aren't well-conditioned. That must be fixed by https://github.com/SleipnirGroup/Sleipnir/issues/483.

Basically, implementing this feature alone won't fix your problem.

laviRZ, 2024-10-13:
> ...so either the duals diverge or the solver enters feasibility restoration more often. Dual unboundedness must be fixed by https://github.com/SleipnirGroup/Sleipnir/issues/483.

I don't quite understand what's going on here, although,

> Basically, implementing this feature alone won't fix your problem.

Maybe it won't solve it completely, but it would help tremendously (and it seems much easier to implement?). All you need is to play around with Choreo for a few minutes to see this happen. As the path gets longer, the solve time (with no entry zones, yes...) scales horribly, even when the path is split between many stop points - but if each section of it would be in a different path, it would be solved without a problem.

spacey-sooty, 2024-10-16:
> but it would help tremendously

Not for obstacles

laviRZ, 2024-10-16:
Why not?
I can attach an example project if you'd like, but the way things are right now, if you have a trajectory on one side of the field, and a tiny obstacle on the other side not interfering at all, depending on the length of the existing trajectory, it might be unsolvable because of that obstacle.

spacey-sooty, 2024-10-16:
> Why not?
> I can attach an example project if you'd like, but the way things are right now, if you have a trajectory on one side of the field, and a tiny obstacle on the other side not interfering at [...]

Because the issue there is with the way the solver handles the obstacles. Tyler could explain this better but to my knowledge keep out constraints make the problem formulation significantly more complex compared to other constraints. This is what caused the feasibility issues, slow generation etc the fix for that is replacing feasibility restoration with One phase IPM

laviRZ, 2024-10-16:
> the fix for the solvers handling of those problems  is replacing feasibility restoration with One phase IPM

Sure, but as I said:
> Maybe it won't solve it completely, but it would help tremendously

You didn't explain why splitting it won't help... In my experience, it doesn't mitigate the issue completely, but at least makes it solvable.

calcmogul, 2024-10-16:
> You didn't explain why splitting it won't help...

It's a poor workaround for a fundamental issue with the solver. I'd rather fix the solver properly.

laviRZ, 2024-10-17:
> > You didn't explain why splitting it won't help...
> 
> It's a poor workaround for a fundamental issue with the solver (I already explained what that issue is earlier). I'd rather fix the solver properly.

Great but if a half measure that's already planned to be implemented and easier to implement could help in the meantime...

calcmogul, 2024-10-17:
It's not actually planned to be implemented. We have other priorities right now.

shueja, 2024-11-20:
Obsolete by improved workflow for segmented trajectories as separate paths.

calcmogul, 2024-11-20:
Now that we have good tooling for shared endpoints on paths, splitting is less important.

closed 2024-11-20 by calcmogul