SleipnirGroup/Choreo #472: Replace nested if let with if let  chain

bruingineer, 2024-05-07:
Tracking issue to address the todo comment below.
https://github.com/rust-lang/rust/issues/53667

in main.rs, line 40, `open_file_dialog(...)`
```
// TODO: Replace with if-let chains (https://github.com/rust-lang/rfcs/pull/2497)
if let Some(path) = file_path {
    if let Some(dir) = path.parent() {
        if let Some(name) = path.file_name() {
            if let Ok(adjacent_gradle) = contains_build_gradle(Some(dir)).await {
                let _ = app_handle.emit_all(
                    "open-file",
                    OpenFileEventPayload {
                        dir: dir.as_os_str().to_str(),
                        name: name.to_str(),
                        contents: file::read_string(path.clone()).ok().as_deref(),
                        adjacent_gradle,
                    },
                );
            }
        }
    }
}
```

spacey-sooty, 2024-05-22:
I think Rust 2024 edition should have support for this

shueja, 2024-09-24:
After the big Rust refactor, we don't have any cases of `if let` where this would be useful

closed 2024-09-24 by shueja