SleipnirGroup/Choreo #491: Allow uses to enter rotation of a waypoint in degrees

Tom-ne, 2024-06-15:
Hello,
I want to create an auto with a waypoint that has theta of 90 degrees, it is a bit annoying to enter in rads...

calcmogul, 2024-06-15:
Rotations may be even better (e.g., 0.25 rotations), since that seems like what you're trying to describe.

pjreiniger, 2024-06-16:
I would also like degrees. Its how humans think of angles. I'd be fine with it being a setting like how you can use imperial when setting wheel diameter / mass / bumper width / etc.

If you lined up against the side of the subwoofer, you were at +/- 60 degrees. That makes much more sense to me to enter into the text field than `1.0472 radians` or `.1666 rotations`

calcmogul, 2024-06-16:
Degrees are just as arbitrary as rotations.

Tom-ne, 2024-06-16:
maybe, but as a human you think in degrees, not rotations and not radians

calcmogul, 2024-06-16:
> but as a human you think in degrees

I think in turns/rotations, not degrees.

Tom-ne, 2024-06-16:
but in rotations you have mostly the same problem with rads...

calcmogul, 2024-06-16:
Not really. The mistake people make with radians is multiplying the π basis into the number. They're much easier to manage if the π is kept separate (e.g., 90 degrees = 0.5π radians). Using τ=2π as a basis is even more convenient because it operations in rotations (e.g., 90 degrees = 0.25τ = 0.25 rotation).

pjreiniger, 2024-06-16:
I can't put `pi / 6` radians or `1 / 6` rotations in the text box. My kids have to use a calculator / google to do the conversion so we can put that a floating point number in there.

Or, you make the units on the field configurable, and everyone can enter values that feels most intuitive for them.

calcmogul, 2024-06-17:
There's https://github.com/SleipnirGroup/Choreo/pull/262, which hasn't been finished.

shueja, 2024-08-25:
Superseded by #96

closed 2024-08-25 by shueja