SleipnirGroup/Choreo #549: Infeasible trajectories with non-square module configurations

Nsl106, 2024-08-05:
When setting the wheelbase and trackwidth to two different numbers it seems a lot of very simple paths fail to generate. As an example, here the wheelbase is 20" and the trackwidth is 17" and it fails to find a solution (same path obviously works with a normal configuration).
[image]

Here's the .chor file I was using for testing this. Github won't upload .chor files which is why it's .txt
[example.chor.txt](https://github.com/user-attachments/files/16500539/example.chor.txt)

calcmogul, 2024-08-05:
I reproduced the issue with 2024.2.3, but it generates fine on main.

calcmogul, 2024-08-05:
I finished bisecting this issue. It was fixed by #505.

Nsl106, 2024-08-16:
I downloaded the version from [here](https://github.com/SleipnirGroup/Choreo/actions/runs/10406383532) and am having trouble building paths still. Is there something else I need to do to get the version with that fix?

calcmogul, 2024-08-16:
Could you upload another .chor that reproduces the issue with the build on main?

Nsl106, 2024-08-16:
So I downloaded the build from [here](https://github.com/SleipnirGroup/Choreo/actions/runs/10406383532) (it was the latest I could find) and ran the installer. Not sure if there's something else I need to be doing, but none of the paths here will generate.
[choreo.chor.txt](https://github.com/user-attachments/files/16639659/choreo.chor.txt)

Nsl106, 2024-08-16:
Still using the build above, it appears that the 4AndCenter path in that file doesn't generate with both wheelbase and trackwidth set to 19.75".

shueja, 2024-08-25:
This is a reported issue exacerbated by non-rectangular module configurations.

bruingineer, 2024-09-08:
@Nsl106 are you seeing the same issues with `main` now?

Nsl106, 2024-09-09:
It doesn't seem to have problems problem generating anymore. Though it still seems to overshoot stop points as mentioned [here](https://discord.com/channels/975739302933856277/1192238434341896294/1276617484614631424), but that's probably unrelated to this issue.

closed 2024-09-21 by shueja