SleipnirGroup/Choreo #574: [choreo] Greatly reduced performance when running 2025 builds

spacey-sooty, 2024-08-26:
When using the latest 2024 Choreo release most things responded to clicks pretty much instantly, on the 2025 version there is a noticeable delay (roughly 1-2 seconds mostly) when clicking things. This is worst on the configuration panel.
[image]
[image]

shueja, 2024-09-02:
Please confirm if this is still an issue on main.

spacey-sooty, 2024-09-03:
Nope this is way better on main

closed 2024-09-03 by spacey-sooty