SleipnirGroup/Choreo #608: [choreo] The dotted line shown when placing constraints is totally out of place

Gold856, 2024-08-29:
Steps to reproduce:
1. Place down a waypoint.
2. Select a constraint and click on the waypoint.
3. Observe as the dotted line points off to infinity.

shueja, 2024-08-29:
Can't reproduce on Windows, but I know a likely cause. Please respond with system details.

Gold856, 2024-08-29:
I'm using a Windows 11 (23H2) laptop with a trackpad.

shueja, 2024-08-29:
For later investigation: somehow `uiState.fieldMatrix` is not getting set when the field component mounts.

Gold856, 2024-08-29:
This bug seems to be correlated with resizing the Choreo window right at startup. When that happens, the window has a black space in the new area and the UI doesn't fill the screen. An easier way to reproduce is to put your system under load (to the point where typing is laggy, an easy way to do this is to build allwpilib), launch Choreo, and immediately hit Win + Arrow Up to maximize the window.

shueja, 2024-08-29:
Thanks for that research. Is the resizing happening before the Field section appears?

shueja, 2024-08-29:
Would you please try https://github.com/SleipnirGroup/Choreo/actions/runs/10620118907 with your same reproduction steps when you get a chance? I can't reproduce anymore with this change, but maybe I misunderstood your instructions.

Gold856, 2024-08-29:
Yes, the resizing happens before the Field section appears. The build you provided seems to have fixed the bug, since I tried to reproduce it for a while, and I couldn't get it to appear. So I'm ready to close this issue when that gets merged into main. Thanks!

PR #617 (2024-09-02): Fix placement of dotted line shown when placing constraints
Reporter of #608 said this build fixed the issue.

Closes #608
closed 2024-09-02 by shueja