SleipnirGroup/Choreo #609: [choreo] Selecting a waypoint as part of a constraint, then attempting to undo causes the whole screen to go blank

Gold856, 2024-08-29:
Steps to reproduce:
1. Place down a waypoint.
2. Select a constraint, then click once on the waypoint. Do not apply the constraint to the sole waypoint.
3. Undo. Observe as the whole screen goes blank.

Choreo version: f571654eb2f043b85c86264485eea35ad3ba4246
Error snippet from the console:
```
index-LYRFLYvb.js:40  TypeError: Cannot read properties of undefined (reading 'x')
    at HBe.render (index-LYRFLYvb.js:316:112461)
    at zC (index-LYRFLYvb.js:42:6790)
    at index-LYRFLYvb.js:45:651
    at r5 (index-LYRFLYvb.js:42:12991)
    at t.track (index-LYRFLYvb.js:42:17871)
    at HBe.o [as render] (index-LYRFLYvb.js:45:628)
    at tS (index-LYRFLYvb.js:40:6181)
    at eM (index-LYRFLYvb.js:40:5978)
    at I4 (index-LYRFLYvb.js:40:44904)
    at O4 (index-LYRFLYvb.js:40:39777)
Jw @ index-LYRFLYvb.js:40
r.callback @ index-LYRFLYvb.js:40
qA @ index-LYRFLYvb.js:38
oM @ index-LYRFLYvb.js:40
N4 @ index-LYRFLYvb.js:40
sY @ index-LYRFLYvb.js:40
dY @ index-LYRFLYvb.js:40
sl @ index-LYRFLYvb.js:40
lM @ index-LYRFLYvb.js:40
Fu @ index-LYRFLYvb.js:38
```

shueja, 2024-09-02:
I can't reproduce this on latest main.

Gold856, 2024-09-02:
I can still reproduce this on latest main. Here's a stack trace with source info:
```
TypeError: Cannot read properties of undefined (reading 'x')
    at FieldConstraintsAddLayer2.render (FieldConstraintAddLayer.tsx:127:27)
    at allowStateChanges (mobx.esm.js:1318:12)
    at mobxreact.esm.js:355:21
    at trackDerivedFunction (mobx.esm.js:1886:18)
    at Reaction2.track (mobx.esm.js:2477:18)
    at FieldConstraintsAddLayer2.reactiveRender [as render] (mobxreact.esm.js:351:14)
    at jj (react-dom.production.min.js:203:189)
    at hj (react-dom.production.min.js:202:149)
    at Vk (react-dom.production.min.js:291:172)
    at Uk (react-dom.production.min.js:279:389)
```

shueja, 2024-09-02:
Got it, my reproductions were with an opened file as the last step, not an added waypoint.

PR #652 (2024-09-02): Fix constraint add blank screen on undo
Closes #609
closed 2024-09-02 by shueja