SleipnirGroup/Choreo #653: [choreo] Change some logic that wants a .chor to a project directory

oh-yes-0-fps, 2024-09-02:
As of 2025 it only makes sense to have 1 `.chor` per directory as all `.traj` in that dir implicitly belongs to all `.chore` and having one `.traj` belong to more than `.chor` seems unwise.

I propose we change some logic in the GUI (and likely backend) to load a directory instead of a project.

Also maybe adding the ability to (with a separate action) select a robot project to create a project in

spacey-sooty, 2024-09-03:
Counterpoint, what if people want to have multiple projects in the same directory?

oh-yes-0-fps, 2024-09-03:
Then projects or trajectory have to keep track of who owns what, like I said having multiple projects own a single trajectory should not be supported