SleipnirGroup/Choreo #678: Traj files use a 2D array for samples?

Gold856, 2024-09-03:
Why is `samples` a 2D array? Maybe there's a feature in Choreo that uses this, but I couldn't find it if it exists. Regardless, it's a really weird type, and it makes JSON deserialization a bit harder.

oh-yes-0-fps, 2024-09-03:
It was done that way to implement stop points, instead of trajectory groups spanning multiple files each segment of the group is in 1 file. Changing it to a 1d array with indexes at stop points is another option.

Gold856, 2024-09-03:
Stop points _are_ implemented with a 1d array. I just generated a trajectory with a few stop points and I can see the samples where the velocity is zero right next to the other samples, all in the same 1d array. So if stop points don't actually use 2d arrays, can Choreo just use a 1d array instead?

shueja, 2024-09-03:
> It was done that way to implement stop points, instead of trajectory groups spanning multiple files each segment of the group is in 1 file. Changing it to a 1d array with indexes at stop points is another option.

This is not quite true. The 2d array implements split points, which are not necessarily tied to stop points, but are rather tied to the split toggle on the waypoint config panel. The 2024 method created a lot of redundant data in the deploy directory. If you’re trying to make a custom deserializer, we can help with that though.

closed 2024-09-04 by shueja