SleipnirGroup/Choreo #774: Low gear ratios cause "Infeasible Problem"

ThatChair, 2024-09-26:
This year, our robot had a very low gear ratio (3.14 : 1 to be precise). However, when I enter this gear ratio into Choreo, I always get an "Infeasible Problem Detected" and the path will not generate. Our settings are shown below.

[image]

Is there any way to work around or fix this? We would love to use Choreo, but we have just defaulted to telling it our robot weighs 120kg to get it to accelerate slowly enough.

bruingineer, 2024-09-26:
Have you tried using a lower motor torque?

I can look into this issue this week.

bruingineer, 2024-09-28:
@ThatChair would you be able to share your choreo file? You can use the "export diagnostic report" in the menu to generate the zip file
[image]

ThatChair, 2024-10-01:
Sorry for the late reply, but I don't see the diagnostic report button. When I pull up the menu, all I see is this: [image]

calcmogul, 2024-10-01:
That button was added after the 2024.2.3 release.

ThatChair, 2024-10-01:
Okay, here it is (in a zip because Github wont let me upload it):
[testAuto.zip](https://github.com/user-attachments/files/17213527/testAuto.zip)

ThatChair, 2024-10-02:
The specifics seem to work in the new 2025 beta.