SleipnirGroup/Choreo #816: Keep-out and keep-in constraints causing weird paths and path generation when used in combination

MidgleyM, 2024-10-02:
diagnostics: [choreo-diagnostics-EqfN9ShL.zip](https://github.com/user-attachments/files/17235184/choreo-diagnostics-EqfN9ShL.zip)

version: Choreo-2025-beta-1

I tried to see if using a few keep out circles would work all at once with each other, and I got some very weird results. Changing the scope of the constraints changes how the path reacts to it, and using a single, bigger keep-out-circle makes it work. Having some sensible scoping makes the path generate, but in a very abnormal way, unless you turn off the keep-in rectangle for the field, then it generates fine.

calcmogul, 2024-10-02:
That's a known issue with keep-out constraints. The interior-point method Choreo uses doesn't properly manage the constraint duals. A solver rewrite is required to fix it: https://github.com/SleipnirGroup/Sleipnir/issues/483.

calcmogul, 2026-09-05:
The solver upgrades on main seem to have fixed this. The trajectory generates fine with all the keep-in and keep-out constraints enabled.

[image]
[image]

closed 2026-09-05 by calcmogul