SleipnirGroup/Choreo #841: Solver hits local minimum?

Gold856, 2024-10-05:
[choreo-diagnostics-L0jt2wS1.zip](https://github.com/user-attachments/files/17267278/choreo-diagnostics-L0jt2wS1.zip)

This path does two 360s which seems not optimal. The commit before https://github.com/SleipnirGroup/Choreo/commit/f3c2a19fea7701f6afa6485e4c7eeebe35305f0d generated a much better path:
[choreo-diagnostics-SMQUqTpZ.zip](https://github.com/user-attachments/files/17267283/choreo-diagnostics-SMQUqTpZ.zip)

Gold856, 2024-10-05:
I think Choreo might have an issue with generating trajectories on the red alliance. Here's a version with a red and blue alliance path. They should be very similar, yet the red alliance version takes longer to generate and does 1.5 rotations at the start.
[choreo-diagnostics-Lbngs3xO.zip](https://github.com/user-attachments/files/17267617/choreo-diagnostics-Lbngs3xO.zip)

spacey-sooty, 2024-10-06:
This is a regression from beta 2.

bruingineer, 2024-10-07:
made similar behaviour on the blue side trajectory when waypoint 2 has -pi rad heading:
[image]

note: the red behaviour goes away if waypoint 2's heading is changed from -pi rad to pi rad.

PR #847 (2024-10-07): [trajoptlib] fix inputModulus
input modulus incorrectly used a `+` where a `-` was needed

fixes #841
closed 2024-10-07 by calcmogul