SleipnirGroup/Choreo #1049: CLI and GUI still produce different results in beta8

pjreiniger, 2024-12-19:
Same idea as #1006.

It appears in beta8 that the the GUI is writing out `.traj` files with `"version": 1`, while the cli writes out `"version": 0`. There is also a problem with one writing out `0` and the other writing out `-0` in certain situations.

Attached diff from generating with the gui as the base, then running the cli on top
[beta8_example.patch](https://github.com/user-attachments/files/18202748/beta8_example.patch)

calcmogul, 2024-12-19:
Looks like we forgot to update the schema version somewhere.

We could just normalize `-0` to `0`.

pjreiniger, 2024-12-19:
The GUI writes out `version 0`*

This means that you can not deploy / simulate after opening the gui

shueja, 2024-12-20:
Fixed in beta 9

shueja, 2024-12-20:
> We could just normalize `-0` to `0` since they mean the same thing in floating point.

In Rust, `(-0.2f64).round() == -0.0`, so an extra normalization step is a valid thing to add.

shueja, 2024-12-20:
> Attached diff from generating with the gui as the base, then running the cli on top

Can you please clarify which platform the GUI and CLI are running on? Your setup from the other issue sounds like the GUI generates on a personal machine (windows, likely), and the CLI generates in a GitHub runner using a different platform.

We expect slightly different outcomes from the generation on different platforms (thus the rounding/normalization of differences below a given tolerance), but between the CLI and GUI on the same platform, they should be running the same code path and getting the same results.

pjreiniger, 2024-12-21:
At the moment I do this on my Windows 11 machine:
- Download the latest windows `setup.exe` (just tried beta 9), and install it.
- Run the gui, and regenerate every path
- Make a git commit
- Run the CLI
-Observe differences. As mentioned above, the big one now is rounding negative 0.

I know if that doesn't work, there is no hope for it to work in CI.

I would *think* that on different architectures should be able to produce identical values down to millimeter precision. It would be very annoying if my Mac kids cannot run choreo on their computer and have it produce deterministic results compared to our team owned driver station computers.

If needed I can easily run the CI action on windows once that starts to work instead of linux, but that was my plan.

pjreiniger, 2024-12-21:
I tried the last master build. Almost there

[still_bad.patch](https://github.com/user-attachments/files/18217822/still_bad.patch)

I haven't looked in depth at the code, but it still seems odd to me that the gui and cli generate different results. Is the gui version sending waypoints down to rust that might have small floating point differences as opposed to what would get written to the file? The code path is obviously not actually the same.

calcmogul, 2024-12-21:
The number written to the file looks numerically identical but in different representations. Perhaps we could force one or the other based on magnitude.

shueja, 2024-12-30:
This uncovered an issue where the Rust component of Choreo is mutating the parameters and really shouldn't be. #1070 addresses this.