SleipnirGroup/Choreo #1078: [New Season] Steps to Add New Field

shueja, 2024-12-31:
Where needed, use field dimensions from WPILib 2025 AprilTag layout JSON
- [ ] Add a new flipper option (including and change the default year in:
- [ ] * `choreolib/py/choreo/util/__init__.py `
- [ ] * `choreolib/src/main/native/include/choreo/util/AllianceFlipperUtil.h`
- [ ] * `choreolib/src/main/java/choreo/util/ChoreoAllianceFlipUtil.java

- [ ] Add new SVG field image
- [ ] Update the usage of field dimensions to set view size in FieldOverlayRoot
- [ ] Change the default dimensions of the field wall obstacle in PathListStore.addPath().

spacey-sooty, 2024-12-31:
Should we add a docs page with this to contributing?

shueja, 2024-12-31:
After we do the 2025 update and find out whatever is missing from the above list, certainly.