SleipnirGroup/Choreo #1155: Error for not explicitly providing Year parameter to Flipped()

calcmogul, 2025-01-18:
From https://github.com/SleipnirGroup/Choreo/pull/1149#issuecomment-2599973157.

onthegrid007, 2025-01-18:
It is also erroring out on ones where there is no template specifier. I see the changes to provide default template specification args to the function but there are functions like GetInitialSample() which do not even have a template specifier option for the Flipped function it uses within...

onthegrid007, 2025-01-18:
Also when will these changes be pushed to the actual vendorlib used in vscode, I see merged changes to the repo but none of them appear in vscode when using the json repo file for online library... @calcmogul

closed 2025-01-18 by calcmogul

calcmogul, 2025-01-18:
I have no idea when Choreo's next release will be.

calcmogul, 2025-01-18:
> It is also erroring out on ones where there is no template specifier.

ChoreoLib C++ is severely lacking in tests. Rather than playing whack-a-mole, it would probably be faster to just copy-paste ChoreoLib C++ into your codebase, fix the compilation errors, then PR them back up.

onthegrid007, 2025-01-21:
@calcmogul 
I pulled the repo and copied the folder structures directly into my code base and it worked flawlessly. It just seems like the JSON file to pull the repo in WPI-VSCode is severely out of date. Since that does not directly relate to this repository hosted on GitHub I can not help fix that... All errors I had seem to be getting were directly related to the outdated ChoreoLib pull through that online library json file upload. If you can just push a release of the downstream repo here to the JSON file location and all should be well.

PR #1156 (2025-01-18): [choreolib] Set default year for sample flip
Fixes #1155.