SleipnirGroup/Choreo #1157: Resetting odometry and then running a trajectory causes only 2 samples to be processed

Gold856, 2025-01-19:
My team followed https://choreo.autos/choreolib/auto-factory/#using-command-compositions, but when we made a command sequence to reset the odometry and then follow a trajectory, we found that our trajectory follower only got called twice.
```java
public Command leave() {
	return sequence(
			m_factory.resetOdometry("Leave"), // Breaks things?
			m_factory.trajectoryCmd("Leave"));
}
```
After setting up a trajectory logger to print the trajectory name and if it's starting the trajectory, as well as modifying our trajectory following method to print out `Following Trajectory`, we saw this printed in the console:
```
Round Trip
true
Round Trip
false
Following Trajectory
Following Trajectory
```
When we removed the resetOdometry command, the trajectory was followed like it's supposed to be:
```java
public Command leave() { // Works fine
	return sequence(m_factory.trajectoryCmd("Leave"));
}
```
We saw this printed in the console:
```
Round Trip
true
Following Trajectory
// Repeat a bunch more times
Following Trajectory
Round Trip
false
Following Trajectory
Following Trajectory
```
This was only tried in sim on Windows 11, unsure if the real robot would be affected.
Running ChoreoLib v2025.0.1.

spacey-sooty, 2025-01-19:
Can you link your code?

Gold856, 2025-01-19:
MRE here: https://github.com/team20/Robot2025/blob/choreo-mre/src/main/java/frc/robot/CommandComposer.java
https://github.com/team20/Robot2025/blob/choreo-mre/src/main/java/frc/robot/subsystems/DriveSubsystem.java

shueja, 2025-01-20:
Bug fixed by #1160