SleipnirGroup/Choreo #1177: Error while importing chorero trajectory: [json.exception.type_error.302] type must be number, but is null

onthegrid007, 2025-01-23:
We just generated a new trajectory from Choreo but it is throwing this error when we import the file.
This is our import function
Is this an issue with the way we import the traj file, or is it a wpilib json parsing error?
```
void C::ImportTrajectory(const std::string& fileName) {
    std::lock_guard<std::mutex> lock(s_mtx);
    try {
        const auto traj{choreo::Choreo::LoadTrajectory<choreo::SwerveSample>(fileName)};
        if (!traj.has_value()) throw std::runtime_error("No trajectory imported!");
        s_cMap[fileName] = traj.value();
        std::cout << "Successfully imported chorero trajectory: " << fileName << std::endl;
    }
    catch (const std::exception& e) {
        fmt::print("Error while importing chorero trajectory: {}\n", e.what());
    }
}
```
Thank you

onthegrid007, 2025-01-23:
Also here is the trajectory file for refrence:
```
{
  "samples": [
    {
      "x": 2,
      "y": 7,
      "heading": -4.1265107329535965e-32,
      "angularVelocity": -2.3455838499013673e-18,
      "velocityX": 0.3178401644298667,
      "velocityY": -0.006614165990104463,
      "moduleForcesX": [
        0,
        0,
        0,
        0
      ],
      "moduleForcesY": [
        0,
        0,
        0,
        0
      ],
      "timestamp": 0
    },
    {
      "x": 2.048083477637813,
      "y": 6.998999397307442,
      "heading": 2.4618289549560253e-18,
      "angularVelocity": 2.5591332685088062e-17,
      "velocityX": 0.4998398888503349,
      "velocityY": -0.010401517594029441,
      "moduleForcesX": [
        37.83866163356687,
        37.83866163356687,
        37.83866163356687,
        37.83866163356687
      ],
      "moduleForcesY": [
        -0.7874095210029574,
        -0.7874095210029574,
        -0.7874095210029585,
        -0.7874095210029582
      ],
      "timestamp": 0.09619775994350539
    },
    {
      "x": 2.096166956432853,
      "y": 6.997998794590525,
      "heading": 4.597705628488085e-18,
      "angularVelocity": 2.2202977017867643e-17,
      "velocityX": 0.49983990088057173,
      "velocityY": -0.010401517847245056,
      "moduleForcesX": [
        0.000002501146965020088,
        0.00000250114696489353,
        0.000002501146965020088,
        0.00000250114696489353
      ],
      "moduleForcesY": [
        -5.2644804197470885e-8,
        -5.2644804197470885e-8,
        -5.264480407089356e-8,
        -5.264480407089356e-8
      ],
      "timestamp": 0.19239551988701079
    },
    {
      "x": 2.144250435228182,
      "y": 6.996998191873607,
[... 847 lines cut ...]
        0.011148068727743841,
        0.011148068727743984
      ],
      "moduleForcesY": [
        -0.000306360206065951,
        -0.000306360206065951,
        -0.00030636020606609365,
        -0.00030636020606609365
      ],
      "timestamp": 4.122802668318757
    }
  ],
  "eventMarkers": [
    {
      "timestamp": 2.020152958813613,
      "command": {
        "type": "named",
        "data": {
          "name": "StopIntake"
        }
      }
    }
  ]
}
```

onthegrid007, 2025-01-23:
And the choreo file used to generate it:
```
{
    "version": "v0.4",
    "robotConfiguration": {
        "mass": 38.1,
        "rotationalInertia": 4.837,
        "motorMaxTorque": 0.9060773480662982,
        "motorMaxVelocity": 4704,
        "gearing": 5.087,
        "wheelbase": 0.6731,
        "trackWidth": 0.673,
        "bumperLength": 0.9,
        "bumperWidth": 0.9,
        "wheelRadius": 0.037
    },
    "paths": {
        "path1": {
            "waypoints": [
                {
                    "x": 2,
                    "y": 7,
                    "heading": 0,
                    "isInitialGuess": false,
                    "translationConstrained": true,
                    "headingConstrained": true,
                    "controlIntervalCount": 21
                },
                {
                    "x": 3.008349895477295,
                    "y": 6.9899187088012695,
                    "heading": 0,
                    "isInitialGuess": false,
                    "translationConstrained": true,
                    "headingConstrained": true,
                    "controlIntervalCount": 22
                },
                {
                    "x": 1.9948266744613647,
                    "y": 7.028900146484375,
                    "heading": 0,
                    "isInitialGuess": false,
                    "translationConstrained": true,
                    "headingConstrained": true,
                    "controlIntervalCount": 40
                }
            ],
            "trajectory": [
                {
                    "x": 2,
                    "y": 7,
                    "heading": -4.1265107329535965e-32,
                    "angularVelocity": -2.3455838499013673e-18,
                    "velocityX": 0.3178401644298667,
                    "velocityY": -0.006614165990104463,
                    "moduleForcesX": [
                        0,
                        0,
                        0,
                        0
                    ],
                    "moduleForcesY": [
                        0,
                        0,
                        0,
                        0
                    ],
                    "timestamp": 0
                },
                {
[... 957 lines cut ...]
            "defaultControlIntervalCount": 40,
            "usesDefaultFieldObstacles": true,
            "circleObstacles": [],
            "eventMarkers": [
                {
                    "name": "StopIntake",
                    "target": 1,
                    "trajTargetIndex": 1,
                    "targetTimestamp": 2.020152958813613,
                    "offset": 0,
                    "command": {
                        "type": "named",
                        "data": {
                            "name": "StopIntake"
                        }
                    }
                }
            ],
            "isTrajectoryStale": false
        }
    },
    "splitTrajectoriesAtStopPoints": false,
    "usesObstacles": true
}
```

shueja, 2025-01-23:
This is a 2024 Choreo project file and a 2025 library. The two are not compatible and you can't automatically upgrade Choreo projects from 2024 to 2025. Download Choreo 2025.0.2 (Or whatever is latest when you do this) and start a new project.

onthegrid007, 2025-01-23:
My bad, I'll check it out tomorrow, just installed 2025 version, thanks sorry for the issue.

closed 2025-01-23 by onthegrid007