SleipnirGroup/Choreo #1187: Closing Choreo will not save the project

realleo245, 2025-01-31:
It's too easy to accidentally close Choreo and lose any work that was worked on.

Maybe some sort of dialog box on closing if there are unsaved changes?

calcmogul, 2025-01-31:
Choreo already asks to save the project on exit if it hasn't been saved before. Once it has, it autosaves. If you're losing work, that's probably a bug in the autosave feature.

realleo245, 2025-01-31:
Today, I lost variables after I closed Choreo.

realleo245, 2025-01-31:
I use Windows 11 with 2025.0.2

realleo245, 2025-02-04:
After using Choreo for a while, it appears to be a variable-specific issue. Paths are still saved, but variables are lost, even if paths are generated

shueja, 2025-02-04:
So just to clarify:
You create a new Choreo project and save it to a location.
You then add several new variables.
You then close Choreo and see that the variables you added are not saved in the file?

realleo245, 2025-02-04:
It's an existing Choreo project

shueja, 2025-02-04:
Got it. I'm having trouble reproducing the behavior. 

Do changes to the robot config save correctly?

realleo245, 2025-02-06:
[choreo-diagnostics-CllebGzD.zip](https://github.com/user-attachments/files/18684134/choreo-diagnostics-CllebGzD.zip)

I have difficulty reproducing the exact issue. However, here's the diagnostic zip. I was trying to create pose variables of the entire reef. You can see that the .traj is storing a variable L with some values, but the .chor does not have the variable L.

shueja, 2025-02-06:
Have you ever lost expression variables, or just poses?

realleo245, 2025-02-06:
I've also lost an angle variable

shueja, 2025-02-07:
I'm still unable to reproduce your issue unfortunately. I'm just opening the project, adding a pose variable, closing, and reopening. If you manage to reproduce the issue, please let us know.