SleipnirGroup/Choreo #1190: Choreo adds random curves to paths

realleo245, 2025-02-04:
[image]
Sometimes, Choreo adds weird curves in the middle of a path. Adding a point-at constraints makes it take a more normal path

[image]

calcmogul, 2025-02-04:
Could you upload the project with the issue here? There's an "Export Diagnostic Report" button in the menu.

calcmogul, 2025-02-04:
OK, I've reproduced the issue on the "Station 1 to A" trajectory with Choreo 2025.0.2 on Linux.

realleo245, 2025-02-04:
That's the trajectory with the issue

calcmogul, 2025-02-04:
Looks like it hit a local minimum. https://github.com/SleipnirGroup/Choreo/pull/1123 doesn't fix it either; it has the same issue, even with the point-at constraint included.

calcmogul, 2025-02-04:
Turning off the keep-in rect fixes it on both 2025.0.2 and that PR.

calcmogul, 2025-02-05:
The duals diverge toward the end of the solve when the keep-in rect is included. This can cause weird issues with the interior-point method's path following and lead to it finding weird local minima.

[duals-with-keep-in-rect.txt](https://github.com/user-attachments/files/18664972/duals-with-keep-in-rect.txt)
[duals-without-keep-in-rect.txt](https://github.com/user-attachments/files/18664973/duals-without-keep-in-rect.txt)

calcmogul, 2026-09-05:
The solver upgrades on main fixed this. The "Station 1 to A" trajectory looks correct with the keep-in rect enabled and the point-at constraint disabled.

[image]
[image]

closed 2026-09-05 by calcmogul