SleipnirGroup/Choreo #1199: Choreo does not respect keep-out circle

realleo245, 2025-02-11:
[image]

Choreo does not respect a keep-out circle in a path, as seen in the screenshot above.

calcmogul, 2025-02-11:
Please upload a project here so we can reproduce it. There's a zip export button in the menu.

realleo245, 2025-02-11:
[choreo-diagnostics-qsGotlju.zip](https://github.com/user-attachments/files/18742294/choreo-diagnostics-qsGotlju.zip)

calcmogul, 2025-02-11:
That project's "Station 1 to B" path seems to be missing the empty waypoint shown in the screenshot.

calcmogul, 2025-02-11:
After making some modifications based on the screenshot, I was able to reproduce the issue:
[choreo-diagnostics-hxF4J2Bm.zip](https://github.com/user-attachments/files/18742417/choreo-diagnostics-hxF4J2Bm.zip)

calcmogul, 2025-02-11:
You only applied the keep-out circle to the 2-3 segment. The solver shrunk that segment to almost nothing so the 3-4 segment could move through that space unencumbered. The solver is allowed to do that because you used an empty waypoint.

closed 2025-02-11 by calcmogul