SleipnirGroup/Choreo #1214: Allow removal of default "Nothing" auto in AutoChooser

ncgears, 2025-03-12:
The ability to remove the "Nothing" auton, or any auton, would be useful.  

The goal here is to allow the override of the name "Nothing", which is defined as a final NONE_NAME.  Because the list is sorted alphabetically, teams have no control over where "Nothing" falls in the list.  It is common that teams may want "Nothing" as the first item.

Being able to do .removeRoutine("Nothing") would allow its replacement.

ex. We use numbers to help sort ours like "00: Nothing", "{10..nn}: Some Auton", "9x: Experimental Autons", but "Nothing" is at the end.

ncgears, 2026-01-07:
This would be a fantastic compromise.  Our "do nothing" auton does exactly that, and we agree that keeping some default is a good safety measure.

shueja, 2026-01-07:
See https://github.com/SleipnirGroup/Choreo/pull/1408/files

ncgears, 2026-01-07:
It passes my sniff test, and looks like it handles backwards compat just fine.